FTC7393/FtcRobotController #6: Rework branch structure

zbrachinara, 2022-07-10:
This fork has very disorganized branches. There are lots of branches in this repo whose purpose is unknown, and/or has been left as a temporary branch and never cleaned up. There are also other branches which may be archival, but whose name does not suggest this. 

A system which replaces the current one must satisfy the following conditions:

* leaves minimal change on the master branch 
* documents itself, preferably on the master branch
* Allows archiving of code from past seasons, *along with the version of evlib used last by that season*
* have a naming scheme which makes the purpose of the branch obvious

Refactoring of the branches currently in the repo is left open for now.

zbrachinara, 2022-07-11:
Here's a potential solution:
* `master`: only has upstream code and an additional github actions file
* `evlib-dev`: used for *active* evlib development. Everything except for the TeamCode module can be modified
* `evlib-release`: used only for verified and ready features of evlib
* `season-[year]-[name]`: used for active in-season development. It receives updates from evlib-release, and only TeamCode can be modified.

Because the directories in the season branches and the `evlib-release` branch are exclusive of each other, merge conflicts should be much less common. The branches are also easily searchable, since they are prefixed by their category.

This doesn't address the location of the branch documentation though. And related to the point on searchability, perhaps the name should come before the year on the season branches?

icanotc, 2022-07-12:
zamn!! that's a great idea!!!

zbrachinara, 2022-09-09:
Been a while, last few tasks are to preserve the old branches somewhere and delete them from this repo