FTC7393/FtcRobotController #8: Add driver button to move Placer to Pre-Collect position

robot256, 2023-02-06:
Gianni wants a button (driver 1 left trigger?) to move the Placer to the Pre-Collect state, so it is in position to grab a cone.

zbrachinara, 2023-02-07:
I notice that `driver2.leftBumper` is also used in this line. Are we going to reassign this trigger or should both of these coexist?
https://github.com/FTC7393/FtcRobotController/blob/ed7e58bc27d514e000cdc76ed71442b962961117/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/PowerPlay/PowerPlayTeleOp.kt#L112

robot256, 2023-02-07:
Since they don't do anything now, we should remove that dead code and refuse them elsewhere.

zbrachinara, 2023-02-07:
Ok that works! For the fetcher slow mode functionality itself, it seems like there's no way to get preset movement any slower unless we define a custom PID or the tele-op temporarily enters a state to smoothly move the set point to the destination (which is then more or less just defining a custom PID). I've tentatively checked it off for now, but if we need preset movement to be slower as well, I think that should be another issue.

robot256, 2023-02-07:
You should get information from Roan on what he wants.  For comparison, the Mecanum drive "slow mode" only affects the scaling of the joystick values. Preset speed might ve fine to stay fast.  LinearSlide class will set the speed of the build-in PID if you change maxCorrectionPower to be less than 1. We do this for the lift.

zbrachinara, 2023-02-07:
Changes are ready for review in #13

robot256, 2023-02-07:
Thanks, these look great!  The only thing I would suggest is to move the PreCollect check to the end of the if-else structure, so it has the lowest priority of all the placer buttons. You don't want to prevent scoring because you accidentally press pre-collect early.

At some point I think it would be a good idea to strip out the four state machine classes that we aren't using in Scorer.kt, for the sake of my scroll wheel. We've figured out better ways to automate and semi-automate them at this point. You can make that a separate issue if you want.

zbrachinara, 2023-02-07:
Cool, change is applied. I'll merge the pr and push the state machine removals directly.

closed 2023-02-07 by zbrachinara