FTC7393/FtcRobotController #14: Sus Lever issue

jvandegriff, 2023-02-08:
(Gianni i typing using jonv account)

A few odd behaviors that have caused problems:
The lever resets to the inside position sometimes when the lift hits the upper/lower limit. 
When it does this. it still treats it as if it were in it's previous state. For instance, if you were doing the first stack preset, then it resets, a single press of the button takes you to the second cone preset instead of the first, since it assumes it was already at the first. The same applies to if it was supposed to be in the ground position and such.
The grabber also sometimes doesn't work and needs to be pressed an extra time to move, might also be attached to the above issue

robot256, 2023-02-08:
leverAdvancement needs to be updated when we send discrete commands to the Lever.  This happens in the safety code, and in the two new functions that we just added (leverToPreset and leverToStackHeight).

Check if grabberAdvancement has the same problem.

The safety code that pauses the fetcher needs to be fixed as well.  It needs to call fetcher.stop() instead of using the last command (which could be a long way away from its current position).  I think the safety code should not change the lever command at all.

robot256, 2023-02-08:
Abhi addressed some of this already, need to check with the drivers how much of the problem remains.

robot256, 2023-02-09:
Drivers said the buttons are fine now.

closed 2023-02-09 by robot256