FTC7393/FtcRobotController #15: Automatic timer to stop cycling

robot256, 2023-02-08:
Add a time check in the cycling loop to abort and park if we run out of time waiting for the mechanisms.  The basic form assumes the scoring steps will complete in the same amount of time each cycle:

1. Based on the parking zone, determine the match time at which parking must start (late parking time).
2. Before starting each `fetchAndScore` operation, add the expected length of the operation to the current match time.  If this exceeds the late parking time, exit the loop.
3. If there is still time to score without fetching, do that.

The advanced form of timeout involves quitting mid-score if one of the mechanisms gets stuck.

1. At several points in the `score` and `scoreAndFetch` subroutines, add match time checks based on how long it will take to complete the rest of the subroutine.  For example, inside any loop that waits for the lift or fetcher, because those can be infinitely long.
2. If the time check shows the remaining steps cannot be completed, perform a "safe abort" operation and return from the subroutine--with a value indicating the abort was performed.
3. The loop checks the return code, and if abort was performed goes straight to park.

robot256, 2023-02-21:
Remember to start by copying the Flow Chart google drawing and modifying it to show where the timeout checks go.  The "wait for..." blocks can have another arrow leaving them labeled "timeout".

TheDeveloper101, 2023-02-25:
Got flowchart fully updated, now code needs to be updated to reflect flowchart

TheDeveloper101, 2023-03-04:
Timers are timing out and need to be fixed/tuned. Could probably add telemetry to notify when a timeout is triggered

robot256, 2023-03-11:
While loops on lines 478 and 495 need to have the same giveUpTime logic and fetcher homing behavior before park.
Wait for Fetcher on line 542 needs to have giveUpTime logic.
Wait for Roadrunner on line 569 needs to have fetcher homing behavior before park.

FalseMSP, 2023-03-11:
TODO: a bit of tuning

closed 2023-03-11 by FalseMSP