FTC7393/FtcRobotController #16: Make FTC dashboard values non-volatile

icanotc, 2023-02-08:
> because the dashboard values are nonvolatile so you kind of have to be constantly copying them into the code in case the robot code crashes and erases them.

That's annoying. 

We can save the variables in a log file and read from it when we boot the robot. 

But, how can we ensure that:
- when we read from the config, would it trigger an update so that it reflects on the dashboard? (Could we start the FTC dashboard after the values are read?)
- Is it possible to read it from a log file?
- Is it reliable to read it from a log file?

And also... 

- Are there better options than reading it from a log file?
- Sus among us. 

Research:

These lines looks very sus

https://github.com/acmerobotics/ftc-dashboard/blob/59c6ff2e4346e69d5b70b7e12c2ab14348b50e94/DashboardCore/src/main/java/com/acmerobotics/dashboard/config/reflection/ReflectionConfig.java#L17-L20

robot256, 2023-02-08:
L17-20 look fine. They are simply skipping any (public) field if it is final ir not static.

The more interesting part is the creatVariableFromField function that is apparently supposed to act recursively on neated objects, but doesn't seem to work at the moment.

icanotc, 2023-02-08:
> L17-20 look fine. They are simply skipping any (public) field if it is final ir not static.

Just gonna have those lines as a starting point, since the documentation says that the dashboard can set the value only if the variable is not final, but is public static. I don't think those lines specifically do the dashboard-related stuff, but the code that uses this class does.