FTC7393/FtcRobotController #17: Make RoadRunner maintain position during cycling

robot256, 2023-02-09:
Something needs to be done to make sure the robot is always in the correct position when we start trying to fetch and score.  There are two problems:

1. By default, FollowTrajectorySequence aborts when it gets within 0.5 inches, 5 degrees, or 0.5 seconds of the expected end of the trajectory.  This isn't always good enough.
2. After it finishes, it does nothing to compensate when we drift around on the field.  Other robots can push us, and the fetcher movement rocks the chassis back and forth.  (It DOES set the motor to "Brake", but nothing else.)

Potential solutions:

1. Command RoadRunner on a brief trajectory back to the scoring position before dropping each cone.
2. Command RoadRunner on a brief trajectory with extra-tight tolerances, so it continues "trying" to keep position during the scoring sequence.
3. Override the RoadRunner inputs and set the motors to RUN_TO_POSITION wherever they are, so they will resist movement.

TheDeveloper101, 2023-02-25:
Update 2/25/23: attempts to hold position with zero velocity, need to tell RoadRunner it is allowed to change velocity if position is off so that it will be more aggressive to fix itself

TheDeveloper101, 2023-03-03:
updated hold pos so that there is logic for it to have a velocity. angular correction is very fast, but positional correction is still too slow

TheDeveloper101, 2023-03-03:
need to continue to update gain, might also want to look into deceleration factor

TheDeveloper101, 2023-03-04:
Got the correct velocity, tuned gain. Now trying to test deadzone

robot256, 2023-03-10:
It oscillates when the fetcher starts moving.  Might need to reduce the gain by a factor of two.  Deadzone (area of zero commanded velocity near target) will only help a little if it's truly overshooting as it returns to position.

robot256, 2023-03-11:
Reduced gain, no longer oscillates.  Sometimes waits a few seconds during the first lineup.  There might be some steady state error or just odometry weirdness contributing to our cone grabbing inconsistency.

closed 2023-03-11 by robot256