FTC7393/FtcRobotController #20: Coordinate Fetcher and Lever movement for vertical motion

robot256, 2023-02-24:
For some portion of the Lever travel, it would be very helpful to coordinate the motion of the Fetcher so that the Grabber travels in a vertical line.  This requires a few steps:

1. Predict the location of the Lever based on the series of commands we give it.  This requires measuring the speed as it goes up and down on each side of the "up" position.
2. Compute the "fetcher offset" using the lever length and angle cosine.
3. Add the fetcher offset to the commanded fetcher position (will be a negative number) and send this as the actual fetcher command.

zbrachinara, 2023-03-19:
[image]
Correction to the diagram: Pictured is a min() function, which should actually be a max() function.

robot256, 2023-03-20:
(Per discord/in person discussion)  Turns out we don't want compensation active all the time.  Only when we give the Scorer a specific command to "raise lower on a stack", and we're already lined up at a stack, and not manually controlling the fetcher at all.  And probably only when the grabber is closed.

Example as used in Teleop to collect the top cone from a stack:
1. Driver presses the Enter Stack button.  Lever moves to Stack 1 position and sets the `InStack` flag in TeleOp to `True`.
2. Driver moves the Fetcher until in grabbing position.
3. Driver closes the Grabber.
4. Driver presses the Toggle Lever button, requesting a move to INSIDE position.  Driver does NOT move the fetcher.
5. PowerPlayTeleOp sees that Grabber is closed and `InStack == true`.  Instead of the normal command, `scorer.lever.goToPreset(INSIDE)`, TeleOp sends a different command, `scorer.raiseLeverFromStack(INSIDE)`.
6. `Scorer.raiseLeverFromStack(INSIDE)` stores the final preset (`finalLeverTarget = INSIDE`), calculates what height the Lever needs to reach to fully unstack a cone from its current position (`coordinateLeverTarget`), and sets the `coordinateFetcher` flag to `true`.  Then it initializes the offset and `requestedFetcherCommand` to begin coordinated movement, and commands the Lever servo to the `coordinateLeverTarget`.
7. `Scorer.act()` sees that `coordinateFetcher == true`, and does the following:  If Lever servo is not done, read current lever position, calculate offset, and send a new fetcher command.  If Lever servo is done reaching `coordinateLeverTarget`, reset the fetcher command do it doesn't move from where it is after offset is removed, and command the Lever servo to the `finalLeverTarget` preset.

In Auto, we would use the same `scorer.raiseLeverFromStack()` function.

robot256, 2023-03-23:
Actually, this all might be easier if we used Vincent's earlier suggestion.  Don't store a separate "requested position", so the joystick always controls where the fetcher goes from where it is right now.  Store "previous offset" and calculate "new offset", and then apply the "delta offset" each tick as though it were a joystick command.  

When you engage offset mode, you set "previous offset" and "new offset" to the offset at the current lever angle, and set the "use offset" flag.  When "use offset" is true, every cycle calculates a new "new offset" and "delta offset".  This means "delta offset" is zero until the Lever moves, and the fetcher doesn't move automatically from where it starts until the Lever moves.

When you disengage offset mode, you set "previous offset" and "new offset" to zero and clear the "use offset" flag.  Now "new offset" doesn't change and "delta offset" is always zero.  No actual command is sent to the fetcher, so it stays right where it was when offset mode was disengaged.

icanotc, 2023-03-24:
**safety feature:** 
stop the fetcher from moving if the lever is still on the stack. 
**assumption:** if the offset flag is set, we are still on the stack.
**implementations:**

Scorer.kt
- [ ] Add flag to enable offset, this could only be through functions under the Scorer class, and we can reasonably infer whether the fetcher is in a stack or not, thus setting the flag as needed.
- [ ] Add `perviousOffset`, `newOffset`, and `deltaOffset`, see above.
- [ ] `raiseLeverFromStack()` This homes the lever while having offsets enabled, does not move the fetcher.
- [ ] Modify other functions that involve the lever so that we can correctly set the `useOffset` flag.
TeleOp:
The only time where the offset is needed is when scoring on stacks.
- [ ] implement toggle stack mode button. click to toggle? click for on and double click for off? TODO: ask drivers
Auto:
We might explicitly set the offset modes here since we can do a lot more complicated and precise controls, the whole point of having offsets in auto was to make coding easy to write and easy to read. (basically)

robot256, 2023-03-24:
> Add perviousOffset, newOffset, and deltaOffset, see above.

Only previousOffset needs to be a class member (state variable). newOffset and deltaOffset will be local variables inside Scorer::act().

icanotc, 2023-03-25:
functions related to lever in the scorer class:
`leverEnterStack` => turn on offset (it moves all 4 non stack presets to level 1
`leverToPreset` => turn on offset except for presets below
`leverToStackHeight` => turn on even for ground preset, since its collecting from the stack still
`leverToggleHome` => turn off the offset
we dont want to have offsets when the state of the grabber is `LeverPresets.INSIDE`, `LeverPresets.GROUND`, `LeverPresets.VERTICAL`, or `LeverPresets.HORIZONTAL` for purposes other than doing it on the stack

icanotc, 2023-03-25:
TODO from today:

- [ ] add a check on whether the grabber is close when pulling off cones from the stack

robot256, 2023-03-25:
Turns out we actually do want the offset to apply to going from GROUND to INSIDE.

I think we got some logic wrong with all the mins and maxes and one-minuses.  The real crux of the issue is that we want to offset the fetcher toward zero when the lever moves from below-horizontal to horizontal, and don't want to apply any offset if it's above horizontal.

ANOTHER ISSUE we might be having is if the fetcher speed is not calibrated correctly, we wont' have the right angle estimate.

icanotc, 2023-03-26:
closed by #41

PR #31 (2023-03-19): Coordinate Fetcher and Lever vertical motion
Closes #20
closed 2023-03-26 by icanotc