FTC7393/FtcRobotController #22: Tune PID for Fetcher motor

robot256, 2023-02-24:


FalseMSP, 2023-03-11:
yes
need more tuning
(probably integral)
proportional has been tuned

robot256, 2023-03-20:
Fetcher is pretty well tuned at velocity PIDF = (30, 1, 0, 0).  The reason it was getting stuck in auto is because the lever compensation code was not accounted for.

robot256, 2023-03-29:
There is a lot of issues with integrator windup when going out and stopping.  We need to reduce the integral gain and raise the proportional.

robot256, 2023-03-30:
My dudes, we might need to try using the "F" part of PIDF. That might make it jump start the motor early.

robot256, 2023-04-11:
We found good numbers to use in teleop, with no integrator whatsoever.  We need to find better numbers for auto that don't have 25 counts of steady state error.

closed 2023-08-05 by robot256