FTC7393/FtcRobotController #23: Dropper Limit Switch

electricPotatoChip, 2023-03-02:
Gianni finally made an account

We just installed a limit switch on the Dropper to detect when a cone has been fully ingested. We should set this up for use, at least telemetry to see if it works reliably. If it does, then we can change some things:

RB changes the state of the Dropper when pressed instead of held. With one press, it will go down (and spin the dropper servo like it currently does when held) until the switch is pressed, then go back to the brace state. Another press will bring it back to the brace state before the switch is pressed, in case you want to bring it back.

We can also integrate this into the Auto once we test it

jvandegriff, 2023-03-10:
Gianni finished wiring the limit switch. It's plugged into port 2, so it's ready to use in the code.

zbrachinara, 2023-03-18:
Status update on the limit switch wiring: We are now no longer sure that the limit switch is wired properly. Wait to test.

robot256, 2023-03-20:
Limit switch can't be pressed on v7 dropper.  v7r2 dropper will be installed tonight.

robot256, 2023-03-20:
Additional comment from #40:

electricPotatoChip:

Dropper limit switch will be ready for use today (Mar20), so we can add it to the tele-op and auto
Tele-op so that when you press RB, it will collect (placer to store and dropper to collect) until the limit switch is pressed, then move back to the brace state and stop intaking.
Pressing RB again during this will cancel the action and return to brace

Adding it to the auto so that it collects until the switch is pressed would also be a good idea

robot256:

What happens if the limit switch doesn't get pressed? Does the dropper spin forever?

robot256, 2023-03-26:
In Teleop, when the collect button is pressed, the dropper should ingest continuously as it does now.
When the collect button is released, the dropper should continue to collect for up to 2 seconds until the dropper limit switch is pressed.

closed 2023-03-26 by TheDeveloper101