FTC7393/FtcRobotController #35: Wait for Fetcher not working in Auto

robot256, 2023-03-20:
With Fetcher Coordination turned on, the fetcher encoder does not match the requested position we send to it from the Auto opmode.  This confused the "wait for fetcher" code and it frequently got stuck.  We need to use a different equation to determine when the Fetcher is in a good position for grabbing.  Either read the actual command back out, or go back to using the `fetcher.isDone` flag.  We can change the margin of error used for the isDone flag if we want.

robot256, 2023-03-20:
Related to #20

robot256, 2023-03-26:
It only hangs on this occasionally.  We could increase the tolerance for the isDone check.

TheDeveloper101, 2023-03-27:
alrighty

robot256, 2023-03-28:
Gonna call this done thanks to 385235a603452122bd2cfcc2859a64d94f1259de
Will make new ticket if different problem/solution is required.

closed 2023-03-28 by robot256