FTC7393/FtcRobotController #40: Distance Sensor in Grabber

electricPotatoChip, 2023-03-20:
We should use the distance sensor at least in the Auto, possibly in the Tele-op as well for an auto-grab (this will be a separate issue if we want to try to add it, focusing on setting it up for the auto currently)

We can have it so that it will measure the distance on the first grab, then adjust all fetcher values by the same amount

robot256, 2023-03-20:
My comment from #29 :
Step 1: Add to robot config and Scorer.kt. make sure it's not updated unless requested by something.
Step 2: In PowerPlayTeleOp add telemetry and logging of the sensor values. Make sure to only read the sensor once and store the value in both places.
Step 3: Analyze the data to see if there is a reliable comparison to know when a cone is either grabbed or ready to grab.

robot256, 2023-03-27:
This sensor was swapped to a REV 2m distance sensor.

robot256, 2023-04-05:
Now the sensor is changing again to a potentiometer in the finger to tell when it has finished grabbing a cone, or if it got stuck on the wall.

closed 2023-08-05 by robot256