FTC7393/FtcRobotController #56: Fetcher stops too suddenly when joystick released

robot256, 2023-03-29:
The naive algorithm to capture the encoder value when the joystick is released results in a non-intuitive overshoot and back-up jerking motion.  The solution is to tell it to stop moving again when its velocity is near zero, so that it slows down quickly but doesn't back up.

We tried this today using the motor's getVelocity method, and treating values smaller than 200 in magnitude as the "stop again" cue.  This had the interesting effect of setting the setpoint to roughly the point where it stops decelerating, but then it still goes backwards a little bit as the integrator unwinds (it had built up trying to slow down the motor faster).

I think the only way around this is to switch to speed mode for the "first" stop--to intentionally clear the integrator--and then switch back to position mode in the "stop again" code.

There is additional work needed to make sure the "stop again" logic doesn't interfere with additional commands given before it can execute.  If the joystick is commanding positive or negative travel, or if a preset/home command is given, stopAgain should be set to false.  The stopAgain action code should be moved to after the joystick code, so that the joystick has a chance to cancel it in the same loop.

robot256, 2023-03-29:
On further thought, mode changing is probably a bad idea. The log file showed some inconsistency in the getVelocity measurement, so we can find the velocity ourself using the previous encoder value.

icanotc, 2023-04-10:
Was this removed after tuning the fetcher PID?

robot256, 2023-04-10:
> Was this removed after tuning the fetcher PID?

Not really, it's still jerky. We'll need to come up with a better PID for teleop. But also we need to make the teleop stop behavior give the motor time to decelerate.  For example, if setpoint is 800, encoder is 500, and the joystick is just released, set the setpoint to 550 immediately.  (Right now we set it to 500 immediately, and then to some other number a few ms later)

robot256, 2023-04-11:
This can be considered fixed! The PID tuning for teleop isn't as accurate as auto expects, but it lets the driver achieve predictable results when combined with the adjustable "stopping distance" option.

closed 2023-04-11 by robot256