FTC7393/FtcRobotController #58: Log File In Auto

robot256, 2023-03-30:
This is getting ridiculous, we need a log file in auto ASAP. Can someone please make the SparkyMecanumDrive.update() function we call in all the autoOp wait loops do a log file entry at the end of it?

robot256, 2023-03-30:
Autocomplete context informed me that Qualcomm has their own DataLogger class now.  It's very simple.  I tested an implementation.  

https://github.com/FTC7393/FtcRobotController/commit/7519214ea58c25d66234c1ce2c44a9e6759d10dd

A couple other things from that commit are useful too, like making `SparkyMecanumDrive drive` a member of the opmode so we don't have to pass it as arguments.

robot256, 2023-03-30:
Then we need Roadrunner pose error, Lift (setpoint, encoder, limit switch), Fetcher (setpoint, encoder, limit switch, useOffset, pauseFetcher), Curent position for servos Placer, Dropper, Lever, Grabber, Dropper limit switch.

TheDeveloper101, 2023-03-31:
currently working on adding logging everywhere

TheDeveloper101, 2023-03-31:
finally finished omg

closed 2023-03-31 by TheDeveloper101