FTC7393/FtcRobotController #71: Tune auto on all 4 sides

icanotc, 2023-04-08:
the right-side auto does not work for both blue and red.

the left-side auto worked before we installed the guards, but it does not work anymore because its been pushed back by the ground junction.

icanotc, 2023-04-08:
also see #70 and #69 for more info, it should be done together

robot256, 2023-04-12:
Need to test each path individually:

- [ ] Right side High junction
- [ ] Rught side Park only
- [ ] Left side High junction
- [ ] Left side Park only
- [ ] Right side Medium junction
- [ ] Left side Medium junction

icanotc, 2023-04-15:
test these on one side, since the code are the same for each side
- [ ] no bonus fetch
- [ ] less than 5 cones (check this if we don't have that as an option)
test on all sides
- [ ] park only

icanotc, 2023-04-15:
first priority is testing the left side medium

robot256, 2023-04-15:
Don't forget "preload only" (score 1 / no bonus fetch)

FalseMSP, 2023-04-28:
fr need to tune ground junction auto too!!

fr tho
season over
closing issue

closed 2023-04-28 by FalseMSP